arkedge/c2a-core #7: minimum_user, 2nd_obc_user の rename

meltingrabbit, 2023-07-18:
## 詳細
- わかりにくい
- 2nd のように，先頭数字は使いにくい

## close条件
- [ ] ディレクトリやドキュメントのりネーム
- [ ] 各所接頭語などのりネーム

meltingrabbit, 2023-07-18:
@sksat 

- minimum_user → mobc_user_sample
- 2nd_obc_user → subobc_user_sample

でいかがでしょうか？

今 AOBC となってるところも， SUBOBC にrename してしまおうか，とも思ってます．

sksat, 2023-07-19:
Examples みたいなディレクトリの下に入れるので sample みたいな語は入れなくていいと思います。なんなら user も要らなくて、`Examples/mobc` とかでよいのでは。

sksat, 2023-07-19:
2nd obc user の名前があんまりしっくりこないのは目的があんまり明らかでないのが原因かな。まあ一旦 subobc でもそんなに悪くはない（2nd よりはよい）。

meltingrabbit, 2023-07-19:
なるほどね。
そうしますか。

あと、
Examples
はそのままでいいかね？ Samples のほうがいい？

meltingrabbit, 2023-07-19:
あとは、

mobc_sample ではなく mobc になった場合、crate 名的にも大丈夫？（なんか、名前がでかすぎる、とかないよね？ 2nd_obc が、crate 名にできない、みたいな話もあったので、確認したい。）

sksat, 2023-07-19:
sample よりは example 寄り（これは単にそっちの方がよくあるかなというだけ）

meltingrabbit, 2023-07-19:
話それるけど、ディレクトリ名、キャメル or スネーク問題もあったりする？

sksat, 2023-07-19:
ややある。できればディレクトリ名はスネークケースに寄せたい。これはファイル名・ディレクトリ名で大文字か小文字かにあまり意味を持たせたくない（し、Win だとこれは実際機能してなかった）半分、Rust の慣習に寄せたい半分です。

meltingrabbit, 2023-07-19:
影響範囲でかすぎてむずいが，やるならいまか．．． < ディレクトリ名のスネーク化

sksat, 2023-07-19:
そう思う。まあそんなに急ぎでもないので v5 とかに分けてもいいですが、分ける意味がそんなにあるかというと薄い。

sksat, 2023-07-19:
crate 名の話スルーしてた。ディレクトリ名と crate 名は一致する必要ないんで問題ないです。crate 名は `c2a-core-example-mobc` とかにすればよい。

meltingrabbit, 2023-07-19:
了解

では

- minimum_user → mobc
- 2nd_obc_user → subobc

で．

なお，ディレクトリrenameと，ドキュメントの修正は別PRでいきます

sksat, 2023-07-19:
OK