arkedge/c2a-core #10: ディレクトリ名をスネークケースにする

meltingrabbit, 2023-07-19:
## 詳細
- https://github.com/arkedge/c2a-core/issues/7#issuecomment-1641294759 ででた
- 影響範囲が大きいので，精査してやる

## close条件
できたら

meltingrabbit, 2023-07-19:
- https://github.com/arkedge/c2a-core/issues/8 の前にやらないと死にそう

sksat, 2023-07-20:
正確には，snake_case にしたいというよりは現状の機能していないことすらある中途半端な CamelCase をやめたい

sksat, 2023-07-26:
> * [C2A 用語のリネーム #8](https://github.com/arkedge/c2a-core/issues/8) の前にやらないと死にそう

これは別に同時にやればよいし，その方がよい

sksat, 2023-08-03:
@meltingrabbit #8 とも関連しますが，`TlmCmd` って素朴に `tlm_cmd` にしますか？それとも `tlmcmd` という1語ということにしますか？（どうせ略語なのであまり区切る意味がない + https://github.com/arkedge/c2a-tlmcmddb/ とかとの整合）

meltingrabbit, 2023-08-03:
これわからんな．．．．ちょっと F' とか kubos とかどうなってるか調べてほしい @sksat

sksat, 2023-08-03:
他がどうなってるかというよりは C2A というエコシステムの中でどうするかという決めの問題だと思ってます

meltingrabbit, 2023-08-03:
宇宙業界で一般的なほうに寄せたほうが良くないですか？
日本語でいう「テレコマ」に相当する英語がどうなってるか，だと思ってます．

meltingrabbit, 2023-08-03:
まあ，C2A上ではtlmとcmdは区別されてるので，C2Aエコシステムでは，であれば `tlm_cmd` かな．

meltingrabbit, 2023-08-03:
F' は tlm と cmd でフォルダ別れてる．．．

`tlm_cmd` で行きましょう．

meltingrabbit, 2023-08-03:
CTCP; common tlm cmd packet などもあるから，分ける，で確定で．

sksat, 2023-08-03:
本当は tlm と cmd と共通部分（CTCP）でそれぞれ区切りたいやつでしょうねえ。ここでは一旦 tlm_cmd でよさそう

PR #167 (2023-10-26): Rename user Settings dir
## 概要
`src/src_user/Settings` ディレクトリを snake_case に rename する

## Issue
- Resolve #10
PR #168 (2023-10-26): Apply rename migration: user Settings dir
## 概要
#167 の migration を example user に適用する

## Issue / PR
- Resolve #10 
- #167 

## 詳細
- [x] examples/mobc
- [x] examples/subobc

## 検証結果
CI が通ればよし
closed 2023-10-26 by sksat