arkedge/c2a-core #22: rename & refactor: IF list -> DS handler registry

sksat, 2023-07-21:
この rename によって `src_user/IfWrapper/if_list.c` の歪さが表出する．そもそもこの存在をどうにかしたいというのはありつつ，実態に即した場所と名前は `src_user/Settings/DriverSuper/handler_registry.h` などなので，これも変更する必要がある．

sksat, 2023-10-24:
各ハンドラは `HAL_init()`, `HAL_rx()`, `HAL_tx()`, `HAL_reopen()` とかかなあ．場所は一旦 `src/src_user/settings/component_driver_super/hal_handler.h` かな

sksat, 2023-10-24:
`IF_init` とかは実際は関数ポインタの array なので，名前で明示したい（そもそも関数ポインタを集約する構造体を作ってそれの array にすべきではある）

meltingrabbit, 2023-10-24:
handlerだと，EHとかぶるので
hal_handler_registry.h
かなぁ？

meltingrabbit, 2023-10-24:
HAL_init_list
HAL_rx_list

..

でよさそう

meltingrabbit, 2023-10-24:
```
extern  int     (*IF_TX[]    )(void* my_if, void* data_v, int data_size);
```
↓
extern  int     (*HAL_tx_list[]    )(void* config, void* data, int data_size);

meltingrabbit, 2023-10-24:
```
typedef enum
{
  CCSDS,
  UART,
  IF_LIST_MAX
} IF_LIST_ENUM;
```

↓

```
typedef enum
{
  HAL_HANDLER_LIST_CCSDS,
  HAL_HANDLER_LIST_UART,
  HAL_HANDLER_LIST_MAX
} HAL_HANDLER_LIST;
```

meltingrabbit, 2023-10-24:
```
typedef enum
{
  IF_REOPEN_TLM_DISRUPTION = 100
} IF_REOPEN_REASON;
```
↓

```
typedef enum
{
  HAL_HANDLER_REOPEN_REASON_TLM_DISRUPTION = 100
} HAL_HANDLER_REOPEN_REASON;
```

sksat, 2023-10-24:
`CDS_IF_RX_BUFFER_SIZE` も `CDS_HAL_RX_HANDLER_BUFFER_SIZE` かな

meltingrabbit, 2023-10-24:
あ～，そこまで手を出すと rx_buffer_size_in_if_rx_ なども変わるので，issueきって後回し，はいかがだろうか？（だいぶ大変になっちゃう）

meltingrabbit, 2023-10-24:
CDS_IF_RX_BUFFER_SIZE , rx_buffer_size_in_if_rx_  だけだからやっちゃうか

sksat, 2023-10-24:
そう．これだけだし `CDS_IF_RX_BUFFER_SIZE` は core でデフォルト設定提供して（例によって undef して）user 側で定義することもあるやつですよね（つまり core -> user のインターフェースですよね）．なのでこのタイミングで変えたい．

meltingrabbit, 2023-10-24:
OK

meltingrabbit, 2023-10-25:
`ret_from_if_rx` も

meltingrabbit, 2023-10-25:
`ret_from_if_tx` も

meltingrabbit, 2023-10-25:
if_config → hal_config

meltingrabbit, 2023-10-25:
CDS_if_rx_buffer_

meltingrabbit, 2023-10-25:
大文字 `IF_` で grep すると結構出てくるね．コメントが

meltingrabbit, 2023-10-25:
`p_super->interface` を `p_super->hal`  とかにします？

sksat, 2023-10-25:
if_list に意識が引っ張られてたけど，hal_handlers だな

PR #165 (2023-10-26): Rename & Refactor IF list to CDS handler registry
## 概要
`src/src_user/hal/if_list.c` （旧 `src/src_user/IfWrapper/if_list.c`）周りを rename する

## Issue
- #8 
- Resolve #22 

## 詳細
- [x] 各 HAL が提供し CDS から呼ばれる関数群を「HAL handler」と呼称するように変更
  - [x] `IF_REOPEN_REASON` -> `HAL_HANDLER_REOPEN_REASON`
- [x] CDS の HAL handler 関係の設定・メンバの rename
  - [x] `CDS_IF_RX_BUFFER_SIZE` -> `CDS_HAL_RX_BUFFER_SIZE`
  - [x] `CDS_RecStatus->ret_from_if_rx` -> `CDS_RecStatus->ret_from_hal_rx`
  - [x] `CDS_StreamSendStatus->ret_from_if_tx` -> `CDS_StreamSendStatus->ret_from_hal_tx`
  - [x] `CDS_Config->settings->rx_buffer_siz [... cut]
PR #166 (2023-10-26): Apply rename migration: IF list
## 概要
#165 の migration を example user に適用する

## Issue / PR
- #8 
- Resolve #22 
- #165 

## 詳細
- [x] examples/mobc
- [x] examples/subobc

## 検証結果
test へのリンクや，検証結果へのリンク

## 影響範囲
- IF list を参照していた部分
- `ret_from_hal_rx` などを参照していた部分（TlmCmd DB csv を含む）

## 補足
beta release したい