arkedge/c2a-core #28: 歪なディレクトリ構造を修正する

sksat, 2023-07-26:
- あくまで「とりあえず今おかしいモノを修正/マシにする」というモチベーション（ #8 と同じ）
- 現状の設計に対応していないディレクトリ構造を修正するにとどめる（ファイルの分割などはしない）
- コード上の構造（アーキテクチャ）を変更するわけではないが，字面的にも人間にとっても breaking であり，#8 との関連性も強いため，v4 でやるのが適切
- 見た目上の作業及び受け取り方としては異なるので別 Issue とするが，#8 とモチベーションは同一であり，不可分なところも多い
  - 例えば，`Driver` などは一つの語が複数の責務を抱え込んでしまっているため，現状のアーキテクチャに即した rename のみをするとしても `Drivers/Super` のようなディレクトリ構造は消える可能性が高い（`hoge/super` にするうまみは無い）

## 詳細
- System
  - 全部フラットに top dir にいてもいいかもしれない
  - ディレクトリ構造を変えるのではなく，単に `kernel` とかに rename するのでもアリ
- Library
  - `Library` という区切りに意味がなさすぎるので，top dir にあってよい
  - 存在と命名が歪なだけなので，一旦 `utils` とかに rename するのでもアリ
- Script
  - `Script/CI` とか要らない（そもそも手元で回してもいいはずで，目的が CI なのではない）
- C2A service（Application）に載ったモノの整理
  - C2A service という枠組みに載っている（目的が大幅に異なる）モノがとても多いので，枠組みに載ったモノすべてを突っ込むディレクトリを切るのは関心事項に対して不当にディレクトリ構造が煩雑になるだけ
  - 枠組みに載っていることを明示するとしても，`component_service` のように，ディレクトリ構造ではなく名前の一部として示す
  - 関心事項と枠組みに載っているかどうかは別軸の評価なので，木構造で表現するのは無理があるし，前者の方が重要なのでそれに即したディレクトリ構造にする
- Drivers
  - driver（super）は名前から何も伝わらないし，`Drivers/Super` と `Drivers/Protocol` はあまり並列していない

sksat, 2023-07-26:
ただ，c2a-core update はもうしばらくはあらゆるモノが breaking であり続けるだろうと思われるので，#8 と比べて v4 で絶対やらないといけないかというと優先順位はやや低い（つまり，実はユーザにとって非自明さが大きい，という変更ほど優先して「お知らせ」としての major update を使う必要がある）

sksat, 2023-07-26:
それはそれとして，現状のディレクトリ構造への雑感: ディレクトリ構造でアーキテクチャを表現できると思いすぎ（実際の関心に対してディレクトリ構造が不当に深い）・関心ではなく目の前の字面で命名/分割をしがち

sksat, 2023-09-05:
`Drivers/Super` と `Drivers/Protocol` はあまり並列していないと思っていたが，元々くっついていたものを雑に分割した結果らしい．実態としては本体とユーティリティ用のメンバ関数といったところか．なのでむしろ rename した上で一つにしてあげる方がよさそう．

sksat, 2023-10-11:
ディレクトリ名を複数形にするのをやめたい！！！！ `src/src_user/Drivers` とかは分かる（中に `Driver` の実装が複数入っているので）けど，`src/src_core/Drivers` はおかしい．そもそも複数実装があったら複数形にすべき，と思っていたのであれば `IfWrapper`（現 `HAL`）も `IfWrappers`  であるべきだったはずだし，単に表記揺れの類では？

sksat, 2023-10-11:
`applications` は core も user も真に複数の実装が入っている（概念の定義は `system` と `tlm_cmd` にある）ので許せる

sksat, 2023-10-12:
ディレクトリ名は全部単数形に揃えることにした

sksat, 2023-10-12:
`<C2A user>/src/src_user/Test/test` は要らないでしょ

sksat, 2023-10-12:
> ディレクトリ名は全部単数形に揃えることにした

cargo も `examples`, `tests` はあるなあとなっている．全統一よりは core, user で同じディレクトリ名にしようとしたかったことに引き摺られて core のディレクトリが不当に複数形名になっている（中に同じ型の似たようなモノが並んでいるわけではない）やつのみでいいかも（基準は曖昧になるけど）．つまり `Drivers` のみ．
https://doc.rust-lang.org/cargo/guide/project-layout.html

meltingrabbit, 2023-10-12:
OK

closed 2023-10-30 by sksat