arkedge/c2a-core #104: c2a-core repository に吸収合体したツールなどの置き場所を考えたい

meltingrabbit, 2023-09-27:
## 詳細
- https://github.com/arkedge/c2a-core/pull/99 ではひとまず（他よりマシという意味で） root においてる
- `/Tools` ? とかにまとめたいきもちもあり

## close条件
検討して整理したら

sksat, 2023-10-05:
tools というよりは utils といった雰囲気

meltingrabbit, 2023-10-12:
なるほどね．
なんか，

src配下にあるのはどうなの？みたいな気持ちもあるよね（src配下はFSWを起きたい気持ちがある． src/{src_core / user} 構造が改善されればなんとかなる？）

meltingrabbit, 2023-10-12:
と言う意味では，`/sils-runtime` も同様に移動させたいな

sksat, 2023-10-12:
`src` 配下は絶対にナシなのでディレクトリを作るかトップディレクトリに置くかで，ディレクトリを作るのは割と慎重にやりたいので一旦トップディレクトリ，というかんじです

sksat, 2023-10-12:
`sils-runtime` は tools と意図違うので分けたい（なので，「同様に移動」はしたくない）

meltingrabbit, 2023-10-12:
今のrootのを，旧来のc2a-coreのファイルをsrc，それ以外を別フォルダ，にひとまず突っ込むのってなし？
今user←→coreのincludeは絶対パスなので，誰も困らないと思ってます．

sksat, 2023-10-12:
それは絶対にナシです．特にメリットが無い（何度でも言いますが core と user で同じディレクトリによるツリー構造にしておくとうれしいような気がする幻想を捨ててください）上に，C2A user での `src/{src_user, src_core}` と中途半端に衝突して confusing です．

sksat, 2023-10-12:
もしも長期的（v4 以降の breaking change を挟んでも）に `src/{src_user, src_core}` の構造を維持しないといけない事情があるのであれば `c2a-core/src_core` であればナシではないかな，とは思いますが，それよりは `src/{src_user, src_core}` の構造をどうにかする方が優先度が高いので結局微妙かな．

meltingrabbit, 2023-10-12:
ん，伝わってなさそう．今 `src/{src_user, src_core}` の構造は言及してないです（coreのrepositoryの中の話をしているつもり）

sksat, 2023-10-12:
伝わってます．その上で困ります．

meltingrabbit, 2023-10-12:
本repositoryを

- FSWになる（おもに C / Rustの）コード郡．組み込み環境でビルドされるファイルたち
- ツール，util類（吸収合併したpythonのたぐいやSILSのたぐい）

に分けたくないですか？という話のつもりで，

-  core と user で同じディレクトリによるツリー構造にしておく
-  src/{src_user, src_core} の構造を維持しないといけない

などは全く言ってないです（そうした方がいいともいってない）

meltingrabbit, 2023-10-12:
↑直接話した．
https://github.com/ut-issl/c2a-core/issues/6#issuecomment-1514933649 の話が v4.0.0 ではなんともならないので
`/src/c2a-core の FSW のさまざまなディレクトリ` という構造は厳しい（user への影響がたくさんある），とのこと．

sksat, 2023-10-12:
今までの事情とかをあんまり考えずに自然な構成を考えると，こんなかんじになる気がする
- c2a-core
  - tlm_cmd
    - code-generator
  - library
    - sils-runtime
  - utils
    - enum-loader

sksat, 2023-10-13:
適切な置き場の前にリファクタが必要なやつなので， v4.0.0 ではやらない