arkedge/c2a-core #156: v4.0.0-alpha.0 のリリース

sksat, 2023-10-17:
- v3 -> v4 の変更はほとんどが migration script による移行が可能（主目的が rename であり，v4.0.0 での変更はそれに徹するため）
- しかし， #18, #19 に関してはそうではない
- そのため，実際に C2A user で v4 系への core update を行う際，通常の c2a-core update のような作業と migration script + α の作業が発生してしまう．この作業を分割したい（そうでないとレビューも行いにくい）．
- そこで，後出しではあるが #19 のマージ時点でバージョンを切りたい

meltingrabbit, 2023-10-23:
これはcargo publshもC"A_VER_HOGEも設定できない，アドホック対応なので，今回だけとする．
リリースノートを丁寧に書くことで，この対応をする．

というのでどうだろう？

sksat, 2023-10-30:
v4.0.0-alpha.0, v4.0.0-alpha.1, v4.0.0-alpha.2 が発生する
- #157

sksat, 2023-11-10:
した
- https://github.com/arkedge/c2a-core/releases/tag/v4.0.0-alpha.0
- https://github.com/arkedge/c2a-core/releases/tag/v4.0.0-alpha.1
- https://github.com/arkedge/c2a-core/releases/tag/v4.0.0-alpha.2

closed 2023-11-10 by sksat