arkedge/c2a-core #184: リリースフローをどうにかする

sksat, 2023-11-06:
- 現在のリリースルールでは，中途半端な Git flow のような状態になっている
  - `main`, `develop` のブランチが存在するが，`main` の更新は必ず `develop` からのマージで行われることを想定したルールになっている
  - つまり，通常のリリースだけ Git flow 的な形式を有しているが，hotfix が存在しない

修整案は以下
- 真に git-flow をやるための hotfix のためのルールを追加する
- `main` ブランチ（の役割を持つもの）を消す
  - 実際には，今の `develop` を `main` に rename して，`main` で開発もリリースもする，かも
  - 現在のリリース体制では `main` ブランチには存在価値が無い
- `main` ブランチ（の役割を持つもの）を消した上で，過去のバージョンをメンテするためのブランチ・ルールを追加する
  - `stable-4.1` のようなブランチを生やして過去のバージョンのメンテ（patch update）を可能にする
  - これまで c2a-core で hotfix / patch update が行われないがちだったのは，そもそも not breaking な対応が行いにくい構造であるのが半分，過去のバージョンをメンテする仕組みがなかったのが半分

meltingrabbit, 2023-11-07:
> 真に git-flow をやるための hotfix のためのルールを追加する

OK

> main ブランチ（の役割を持つもの）を消す

git flow だとmain (master) 必要ではないだろうか？（もともとの定義でも）
（あと，main で git log --first-parent すると，バージョンごとの差分も見えたりする？）

sksat, 2023-11-07:
たぶん誤読してると思うんですが，3案ある

sksat, 2023-11-07:
ちょっと分かりにくくはあったな

meltingrabbit, 2023-11-07:
あ～，理解．（こういうときは ol つかうとよい？）

meltingrabbit, 2023-11-07:
> 真に git-flow をやるための hotfix のためのルールを追加する

でよいのではないか，とおもいつつも， `stable-4.1 のようなブランチを生やして過去のバージョンのメンテ（patch update）を可能にする` は持たせたいな，という気持ち．

meltingrabbit, 2023-11-07:
main branch はのこしつつ， `main ブランチ（の役割を持つもの）を消した上で，過去のバージョンをメンテするためのブランチ・ルールを追加する` でよいのでは？（main branch は今まで通り．で，hot fix や 軽微なupdateしたいときは，たとえば v3.10.0 に「stable-3.10 のようなブランチを生やして」いく，みたいな？

sksat, 2023-11-07:
いや，それなら main branch まったく要らなくないですか（現時点でも要らないけど，さらに）

sksat, 2023-11-09:
- 嘘 git-flow はやめる
- 過去のバージョンをメンテするためのブランチを切るようにする

sksat, 2023-12-11:
ついに develop branch を葬った

closed 2023-12-11 by sksat