arkedge/c2a-core #216: TlmCmd DB v3 のリリース

meltingrabbit, 2023-11-21:
## 詳細
TlmCmd DB v3 をだす

背景については， TBW

## close条件
- [ ] 現行最新版 https://github.com/ut-issl/tlm-cmd-db/releases/tag/v2.4.1 を c2a-core にとりこむ（monorepo） @sksat 
- [ ] v3 のアップデートとドキュメント整理 @meltingrabbit 
  - https://github.com/ut-issl/tlm-cmd-db/pull/24 相当
  - display info の扱いを，c2a devtoolsに限定することを明記
  - 各種意味論をドキュメントとして整備

meltingrabbit, 2023-11-21:
code generator ，奇跡的に互換性破壊しないことを発見．（じつはWINGSも互換性保たれるような．．．）

meltingrabbit, 2023-11-28:
@sksat そういえば，これモノリポにすると，tlm cmd のバージョンはc2a coreのバージョンと一緒になる，ということよね．
最新の tlm cmd db と c2a user example でつかってるバージョンは常に一致してないとだめ，ということになります？

（過渡期，どうしようかなぁ～～となっていた．Gaia側の対応なども待たないとCI動かない，とか）

sksat, 2023-11-28:
@meltingrabbit リポジトリ管理は1つにするけど，TlmCmdDB のバージョンは別途切る，でもよいと思っていましたが，そこらへんの整理付いてないならそもそもリポジトリ併合するかどうかも込みでちゃんと整理したいです．

meltingrabbit, 2023-11-28:
> TlmCmdDB のバージョンは別途切る

これができるとこちらの方が嬉しいんだけど，できますか？（やり方が分かってない）

あと，enum loader や code generator は core とバージョン揃えるべき（＝独自バージョンもたないべき）だとおもっていて，そういうのの整理ってどうすればいいですか？（ぱっとみて，tlm cmd db だけ独自バージョンもつようなモノリポが，どういうふうに見えるか，わかってない / tlm cmd db が c2a core とバージョンが独立であることを，他の user はどうやって気づくようにできるのか，など）

sksat, 2023-12-21:
これは `arkedge/c2a-tlmcmddb` で行う