arkedge/c2a-core #259: TDSP_execute_pl_as_task_listの返り値でまだ実行すべきタスクがあるかどうかを返したい

KOBA789, 2023-12-26:
## 詳細
TDSP_execute_pl_as_task_list の返り値を boolean にして、まだ実行すべきタスクがあるかもしれない（偽陽性があっていい）ときに true、タスクリストを空振りしたときに false を返すようにしたい。
こうすると、false が返った場合は次のサイクル(tick)まで CPU を寝かせることができるようになり、省電力化の道が拓ける。
タスクリストの空振りで 1ms の間 CPU を無駄にぶん回すのは現代のマイコンでは大変もったいない。
厳密には、C2A_core_main の返り値まで伝播させないといけない。

## close条件
TDSP_execute_pl_as_task_list の返り値を boolean にして、上記のような仕様で返り値を返すようになったら。

## 備考
次の task までの残り step 数を返り値にするという案もあったが、CPU はどうせ 1ms ごとにタイマ割り込みで起きるし、true/false の方がシンプルだろうということになった。

呼び出し側はこんな感じになるのを想定:
```c
void main_loop(void) {
  for (;;) {
    while (C2A_core_main()) { /* noop */ }
    wfi(); // Arm を寝かすやつ
  }
}
```

meltingrabbit, 2023-12-26:
つまり，時刻が進んだ瞬間に再度 `TDSP_execute_pl_as_task_list` を呼ぶようにすることができる（それまでは寝る）

KOBA789, 2024-03-21:
これ、じわっと優先度上げていきたいです。
今後使う CPU が高性能になればなるほど、CPU を空回りさせたときの電力・熱的な問題が大きいためです。

meltingrabbit, 2024-03-21:
PLはO(1)でキュー残量がわかるので，キューに残ってるタスク数（コマンド数）を返却するので良さそう

meltingrabbit, 2024-03-28:
`C2A_core_main` の返り値も変えないとだめそうね．かえます

meltingrabbit, 2024-03-29:
```
void C2A_core_main(void)
{
  TDSP_execute_pl_as_task_list();
  WDT_clear_wdt();
}
```
なので，
`TDSP_execute_pl_as_task_list` を呼び出してから `C2A_core_main` を抜けるまでの間に ti が進んでしまうとだめ．

なので，runtime 側で

```
割り込み停止
if `現在時刻で実行すべき残りタスク数`
  割り込み再開
  wait for interupt
endif
```
みたな構成にする．

@sksat 
FYI ですが，`現在時刻で実行すべき残りタスク数` を返す関数が，runtime 側に公開される関数として追加されそうです．