arkedge/c2a-core #280: NodeJS 系のツールを統一する

sksat, 2024-01-10:


KOBA789, 2024-01-17:
メモ:
- yarn v1 をやめて pnpm に統一する
  - yarn は v2 以降大迷走している
  - npm は遅いし常に壊れていてダメ
  - pnpm はまともだし速い
  - Node.js にバンドルされている corepack を使うと pnpm 自体のバージョン管理ができる
- Node.js は v21 系以降とする
  - corepack のバージョンが新しくてよろしい

- package.json にこのように書いておくと: https://github.com/KOBA789/itzmono-vite/blob/eeea4b00aa436b409ffe0906b81a63ff7dc6efc6/package.json#L51
- `corepack enable` で pnpm が入る: https://github.com/KOBA789/itzmono-vite/blob/eeea4b00aa436b409ffe0906b81a63ff7dc6efc6/README.md#setup

sksat, 2024-04-11:
`tmtc-c2a` のビルドで yarn の依存が生えてしまっているので、v4.4.0 では見送る

KOBA789, 2024-04-16:
tmtc-c2a のビルドも pnpm に寄せなかったっけ？（そういうことではない？）

sksat, 2024-04-16:
それはまだ beta

KOBA789, 2024-04-16:
あ〜〜たしかに。yarn のバージョンを ship したのはミスったな……

KOBA789, 2024-09-30:
corepack が Node.js にバンドルされなくなるのでセットアップの容易さというメリットが失われそう。
いよいよもって cargo-run-bin などを用いて自前のツールチェーンをちゃんと配れるようにすべきと思う。