arkedge/c2a-core #318: BCE Rotator が 0 番目の cmd ではなく 1 番目のコマンドから実行される，らしい

meltingrabbit, 2024-03-08:
## 詳細
基本的には Rotator はブロックコマンド実行周期を分周しているだけなので，どのコマンドから実行してもいいようなロジックを組むのが自然ではあるが，特定ケースでは実行順番を考慮したいケースがある．
その時に 0 番目の Cmd から実行されるのが自然なため，そのように修正する．

ToshiAki64, 2024-03-08:
これ、修正方法は分かっているんですが、（起動時初回のみとはいえ）既存の処理の動作が変わるので、どうしたものかと思ってました（ex. デバッグ出力の出方が変わる）。

meltingrabbit, 2024-03-08:
あ～～

https://github.com/arkedge/c2a-core/blob/3059e598acb252eb931cddddb5dc6ef30eee29f9/examples/mobc/src/src_user/settings/modes/task_lists/elements/tl_elem_debug_display.c#L9-L12

これかぁ．．．
つまり初回に flush screen されるので，init debug msg が消えてしまうんですね．．．

migration note 丁寧に書いて，ここは `AR_APP_DBG_FLUSH_SCREEN` を 9 番目に移動，ですかね．

ToshiAki64, 2024-03-08:
> migration note 丁寧に書いて

なるほど、core update 時の作業のひとつにするって感じですね。

ToshiAki64, 2024-03-28:
マージしたので閉じ

closed 2024-03-28 by ToshiAki64