arkedge/c2a-core #373: SILSでの64bitバイナリの作成方法

mitsut, 2024-09-23:
## 詳細
ホスト環境はWSL Ubuntu 22.04使用しています。
mobc, subobcをcargo build でビルドすると、32bitバイナリが生成されました。
64bitバイナリで作成する方法はありますか？
```
~/c2a-core/examples/subobc$ file ~c2a-core/target/i686-unknown-linux-gnu/debug/c2a-example-subobc
~c2a-core/target/i686-unknown-linux-gnu/debug/c2a-example-subobc: ELF 32-bit LSB pie executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, BuildID[sha1]=941b4605b030619d55409b2720b5675e607dfb05, for GNU/Linux 3.2.0, with debug_info, not stripped
```
Changelogにある情報から CMakeLists.txtやbuild.rsの`C2A_BUILD_FOR_32BIT`をOFF、`-m32` オプションを外すなどを行なってみましたが、変わりませんでした。
## close条件
64bit バイナリの作成方法。もしくは未サポートの回答があれば。

## 備考

sksat, 2024-11-27:
@mitsut すみません。別途ローカルで会話していたこともありこちらで回答できていませんでした。
これは C2A の設定というよりは Cargo の設定の問題です。デフォルトのビルドターゲットは `.cargo/config.toml` で設定されています。`examples/mobc` の場合は https://github.com/arkedge/c2a-core/blob/6b955c7804e5526da9b3b747dae68a5df7389347/examples/mobc/.cargo/config.toml#L2 です。

ただし、この設定はリポジトリ全体でも行っているため、この設定を x86_64 に変える場合は、コメントアウトするのではなく、明示的に `examples/mobc/.cargo/config.toml` で `x86_64-unknown-linux-gnu` などを指定するとよいです。

mitsut, 2025-02-10:
ご回答ありがとうございます

closed 2025-02-10 by mitsut