arkedge/c2a-core #412: `PL_insert_tl_cmd` の失敗などの， core 側の軽微なエラーであっても なしかしらの EL を吐いてほしい

meltingrabbit, 2025-02-12:
## 詳細
core では極力 EL を発火してこなかったが，運用上はたくさん発火したほうが嬉しい（止めることもできるので）ため，core の軽微なエラーであっても EL を発行するようにする

meltingrabbit, 2025-02-14:
PL_insert_tl_cmd は Cmd_TLCD_DEPLOY_BLOCK のほうでは吐かれてるな．
今回はどっちかというと， `PH_analyze_cmd_packet` でのエラー記録っぽそうか．

meltingrabbit, 2025-02-14:
## 議論点
`PH_analyze_cmd_packet` で処理される cmd packet が，キューイング（TL や RT, GS cmd キュー）されるときのエラー（キューがいっぱい（eg, TimeLine cmd list がいっぱい））を EL で記録する方法

### 案１
https://github.com/arkedge/c2a-core/blob/d3ee9e35cf63d5969013c127cdf1b47da6e90a36/tlm_cmd/packet_handler.c#L141-L174

の switch 内部での返り値で EL を発行する．

group: `EL_CORE_GROUP_PH_CMD` , local: { `CCP_EXEC_TYPE_GS` } x { `PH_ACK` }

とする．ロギングの OFF は goup 毎なので， `CCP_EXEC_TYPE_GS` 毎に group id を発行してもいいかもしれない．

メリット
user 側がどんな設計をしても，必ず EL が発行される

デメリット
コマンド種類に関係なくすべて発行される

### 案２

https://github.com/arkedge/c2a-core/blob/d3ee9e35cf63d5969013c127cdf1b47da6e90a36/examples/mobc/src/src_user/component_driver/com/gs.c#L252

で EL を発行する

ドライバレベルでの実装．

メリット
ドライバレベルなので，用途に合わせて group などの設計が可能

デメリット
ドライバは user 側なので，実装されるかはそのドライバ次第

---

@sksat @tarotene 
どう思いますか？
EL は吐き放題なので，まあ案１かなぁとは思ってます．

sksat, 2025-02-14:
案1でよいんではないか

tarotene, 2025-02-14:
案1が良いです

ちなみに「`CCP_EXEC_TYPE_GS` 毎に group id を発行」は結構あると嬉しいですね、検証時に

meltingrabbit, 2025-02-14:
OK

「`CCP_EXEC_TYPE` 毎に group id を発行」すると， `EL_CORE_GROUP` が一気に 10 個ふえるのと， `CCP_EXEC_TYPE` が変わるごとにメンテが必要なのと， `TLCD_ENABLE_MISSION_TL` によって変わるというのがけっこうだるいので，どのぐらい必要か走りたい．

検証時の話だけなのであれば，
local id の u32 のうち，上位 16 bit を `CCP_EXEC_TYPE` ，下位 16 bit を `PH_ACK` にすればいいので，ロジックもシンプルかつ，検証時もそこまで困らないと思うのだけど，いかが？

「`CCP_EXEC_TYPE` 毎に group id を発行」が真に必要なのは，`CCP_EXEC_TYPE` 毎に EL 登録を有効化/無効化したいか？で決めたいです．

@tarotene

tarotene, 2025-02-14:
まさにそれ考えてました

> CCP_EXEC_TYPE 毎に EL 登録を有効化/無効化

する手段がそれしかないのであれば欲しい寄りなんですが、メンテコストの損が大きいならちょっとやり方考えたいですね...（何かハックできないかな...）

meltingrabbit, 2025-02-14:
> CCP_EXEC_TYPE 毎に EL 登録を有効化/無効化する手段がそれしかないのであれば欲しい寄りなんですが、

それしかないですが，ノミナル時はここでエラーは吐かない（EL発行されない（もしここでばんばんエラーでてたらやばい））ので，検証時にコンタミすることはないんじゃないかなぁ，，とは思ってます．

tarotene, 2025-02-14:
なるほど であれば group id の発行は当面は不要として大丈夫そうです

meltingrabbit, 2025-02-14:
了解です
また困ったら教えてください

meltingrabbit, 2025-02-14:
その他

- note は cmd id でよさそう
- PH_user_analyze_cmd のケアは user 側でやってもらうしかないか

closed 2025-02-25 by meltingrabbit