arkedge/c2a-core #444: BCT に書き込み時にセグフォが発生する

hender14, 2025-05-02:
## 詳細
`BCT_save_cmd_` 関数内で const CommonCmdPacket* のパケットを const BCT_CmdData* にキャストさせて、`BCT_set_bc_cmd_data_` 関数を call している

https://github.com/arkedge/c2a-core/blob/d920245ab61784360708c966a8e62b0d7ecd47cc/tlm_cmd/block_command_table.c#L225-L239

その `BCT_set_bc_cmd_data_` 関数では、`sizeof(BCT_CmdData)` のサイズの `value` にある値を `cmd_data` に memcpy しているが、ここの処理を実行時にセグフォが起きてしまっている
(ただし BCT のアドレスによっては、この異常が発生しない)

https://github.com/arkedge/c2a-core/blob/d920245ab61784360708c966a8e62b0d7ecd47cc/tlm_cmd/block_command_table.c#L97-L101

memcpy 時のサイズを `sizeof(BCT_CmdData)` ではなく、`CCP_get_packet_len((const CommonCmdPacket*)value)` に変更すると問題なく動作するため、`cmd_data` の値が 64 Byte 未満になっている可能性が高いと推定している。

原因としては、const CommonCmdPacket* のパケットを const BCT_CmdData* にキャストさせた際に、64 Byte 分のバッファが確保されていない事が推定され、このキャスト時の動作が未定義動作に当たっていたのではないかと考えている

## close条件
対応したら

## 備考

meltingrabbit, 2025-05-07:
普通，

`BCT_CMD_MAX_LENGTH` <= `CCP_MAX_LEN`

のはずだが，ユーザー設定が不適切で

`BCT_CMD_MAX_LENGTH` > `CCP_MAX_LEN`   .... (*)

のときに発生してる？

(*) は本質的にメモリの無駄なので，そういう user settings にならないように，define のアサートを入れる，というのが適切な対応っぽい機がします．

meltingrabbit, 2025-05-14:
これはちょっと根が深い気がするので，ちゃんと見たい（ `BCT_CMD_MAX_LENGTH` あたりの設定がおかしい可能性もあるし，もしそうであれば define のアサートを入れるべきだったり）

あと `CCP_get_packet_len((const CommonCmdPacket*)value)` のキャストは実は NG （今は動くけど， CCP のstruct の構造がかわると動かなくなるので）